jedisct1/libhydrogen #5: Question on Gimli

Scratch-net, 2017-07-10:
Hi and sorry I'm asking here, but did not find a better place to ask dumb questions.

In the spec https://gimli.cr.yp.to/spec.html it is said that the second step in the non-linear layer is
```c
x = x ^ (z << 1) ^ ((y&z) << 2);
y = y ^ x        ^ ((x|z) << 1);
z = z ^ y        ^ ((x&y) << 3);
```
BUT

in the reference code below we see that

```c
      x = rotate(state[    column], 24);
      y = rotate(state[4 + column],  9);
      z =        state[8 + column];

      state[8 + column] = x ^ (z << 1) ^ ((y&z) << 2);
      state[4 + column] = y ^ x        ^ ((x|z) << 1);
      state[column]     = z ^ y        ^ ((x&y) << 3);
```

If you take a look at  
```c
z = state[8 + column];
```
and then 
```c
 state[8 + column] = x ^ (z << 1) ^ ((y&z) << 2);
```

it turns out that actually it's
it's
```c
      z = x ^ (z << 1) ^ ((y&z) << 2);
      y = y ^ x        ^ ((x|z) << 1);
      x = z ^ y        ^ ((x&y) << 3);
```

Am I missing something?

jedisct1, 2017-07-10:
You're missing the swap:
x <- z
z <- x

closed 2017-07-10 by jedisct1

Scratch-net, 2017-07-10:
But the swap step goes after the second step, not before?