jedisct1/libhydrogen #6: Gimli

ildyria, 2017-08-22:
I think your portable implementation of Gimli is wrong.

When using test vectors : 
```
00000000 9e3779ba 3c6ef37a daa66d46 
78dde724 1715611a b54cdb2e 53845566 
f1bbcfc8 8ff34a5a 2e2ac522 cc624026 
```
you get : 
```
248d65cc 93af60b6 d8518d57 8c5c2da7 
15501de2 38f6accb 5ab3d18c b8cea71a 
8ddf7daf 21f4db33 60aec38f 96083e81 
```
instead of : 
```
ba11c85a 91bad119 380ce880 d24c2c68 
3eceffea 277a921c 4f73a0bd da5a9cd8 
84b673f0 34e52ff7 9e2bef49 f41bb8d6 
```

This is due to the fact that you are doing : 
- s.s.s.s for small swap instead of s...s...s...
- .S.S.S.S for big swap instead of ..S...S...S...

jedisct1, 2017-08-22:
This is an intentional tweak introduced to mitigate the slow diffusion of the current Gimli construction (cf. Mike Hamburg's cryptanalysis of 22.5 rounds of Gimli).

It will be replaced by the next version of Gimli, if there is one to address the current weaknesses of the permutation.

jedisct1, 2017-08-22:
Gimli is a wonderful fit for libhydrogen's targets. But even though the way it's used mitigates the attacks described in that paper, these results don't make me feel very comfortable with it in its current form.

ildyria, 2017-09-26:
FYI here is the response from the Gimli team: https://gimli.cr.yp.to/statement.html
Does that makes you more comfortable ?

jedisct1, 2017-09-26:
Fantastic. Thanks a lot for that reassuring statement.

libhydrogen was switched back to the official version. Thanks again!

closed 2017-09-26 by jedisct1