jedisct1/libhydrogen #7: Structure of hydro_secretbox_encrypt output

RandDruid, 2017-09-08:
First of all, thank you for a great library!

I am integrating libhydrogen into my project, and I am noticed one thing which I can't understand.

When hydro_secretbox_encrypt_iv constructs packet, it leaves `gimli_RATE` bytes in the buffer untouched.
`siv` is filled in with 
`mem_cpy(siv, buf + gimli_RATE, hydro_secretbox_SIVBYTES - gimli_RATE);`
and `mac` starts only from `hydro_secretbox_SIVBYTES`.
With default defines it gives 16 unused bytes in every packet.
Is there any reasoning for this?
Is it safe to move `mac` & the rest of the packet 16 bytes to the left to close hole?

jedisct1, 2017-09-08:
Hi,

There are no unused bytes in the packet. `siv` points to the beginning of the packet, `mac`'s offset is `hydro_secretbox_SIVBYTES`, and the actual ciphertext comes right after (`hydro_secretbox_SIVBYTES + hydro_secretbox_MACBYTES`).

During SIV computation, `buf` is used as a temporary state, and `buf + gimli_RATE` is the first byte of the reserve.

RandDruid, 2017-09-08:
Ok, I will take a closer look to computations. In my case bytes between `buf + gimli_RATE` and `buf + hydro_secretbox_SIVBYTES` are not changed by 'encrypt'. I found it the hard way: I didn't clear temporary buffer before encrypt and was surprised by parts of my SQL statements in 'encrypted' traffic :-)

jedisct1, 2017-09-08:
Oops, `hydro_secretbox_SIVBYTES - gimli_RATE` doesn't look correct indeed :/

Fixed in e08349760f00ce457bfb28070c3425768af09fae

Thanks a ton for spotting this.

closed 2017-09-08 by jedisct1