jedisct1/libhydrogen #8: Confusion about signing/key exchanging keys.

opless, 2017-12-17:
Apologies if this is not the correct forum for such a question, directions to a more appropriate forum welcome (I would have posted on stackoverflow, but github seems to have recent activity by the author)

Are both hydro_sign_keypair, and hydro_sign_keypair equivalent ?
I assume both are Curve25519 keys, as their sizes are the same.

Is there some reason they have different names, other than for readability?

Finally, are there any reasons (if they are equivalent) not to share the keypair between both signing and key exchange functions?

jedisct1, 2017-12-17:
Hi,

Both operations indeed leverage the same curve. However:

- Future versions may not share the same curve, or use different key types (especially if a post-quantum signature scheme gets introduced). Having dedicated functions and keys guarantees a stable API.
- `kx_keypair()` "clamps" the secret scalar. The 3 low bits are cleared to make it a multiple of the cofactor, and prevent small subgroup attacks. Since a single party is required for signing, `sign_keypair()` doesn't perform this clamping operation. Using a key exchange key pair for signing will work, but not the other way round.
- This is not specific to the schemes implemented here, but there is no proof that it is safe to use the same key for signatures and for key exchange. These operations are always analyzed independently, and assume that a key is only dedicated to one operation. That being said, there are no known attacks against doing this, and unless the hash function is not preimage resistant, there are few reasons to think that it could not be secure. Still, using dedicated keys remains more reassuring.
- If keys are shared, having a key exchange secret key leaked (possibly due to a bug in the application) means that the attacker will also be able to sign anything. This can have severe implications especially since revoking public keys can be tricky. Different keys can have different time spans, and the compromise of a key type will not affect the other operation.

opless, 2017-12-17:
Understood. 

Thank you for your very timely response 👍

closed 2017-12-17 by opless