jedisct1/libhydrogen #12: PlatformIO library integration

gh0st42, 2018-01-30:
It would be really convenient if this library was listed in the library manager of platformio. As far as I understand, you would only need to add one json file with meta data. More information can be found here:
http://docs.platformio.org/en/latest/librarymanager/creating.html

So far, I am manually cloning this repo, the automatic build system of platformio does the rest and all just works. To reach more users and for automatic dependency management to work an integration into the library manager is still desirable.

Thanks and keep up the great work!

marek22k, 2025-03-04:
> This issue is stale because it has been open for 30 days with no activity.

And that's a reason to close it? Why?

marek22k, 2025-05-04:
> > This issue is stale because it has been open for 30 days with no activity.
> 
> And that's a reason to close it? Why?

Seems the answer is yes.🤷‍♂️

jedisct1, 2025-05-04:
I don't know what PlatformIO is. I can't commit to maintaining something for a platform I don't use, don't know how to test, and frankly don't understand.

If someone who actually uses it could submit a pull request with the necessary files, ideally including a GitHub Actions workflow to automatically test that it works I’d be more than happy to support it.

But if you're just opening an issue saying, "Please add this, even though you don't know anything about it," I don't feel confident enough to improvise.

marek22k, 2025-05-04:
PlatformIO is a kind of framework with which you can program (compile, flash and debug) embedded devices such as Arduinos. As far as I know (it's been a while since I used PlatformIO) the issue refers to the package manager for the PlatformIO library. (this includes for example libraries to address OLED displays and such, but also libraries for other things like cryptography)