jedisct1/libhydrogen #13: Signature Verification Only?

jmbldwn, 2018-02-23:
I have an application that only needs to verify a signature in the embedded system (all signing is done on servers).  It seems to me I could further reduce the footprint if I could prune everything but what is needed for 'hydro_sign_verify'.

A switch for this would be a nice addition, unless I'm missing some other way to easily prune the resulting code.

jedisct1, 2018-02-23:
Hi there!

Indeed, performing only signature verification is a common use case for firmware updates, and it makes sense to have a compilation flag to trim down the code in this scenario.

I'll see what I can do, but in the meantime, you can avoid calling `hydro_init()` (you won't need a PRG for signature verification), and use LTO if your environment supports it. This should get you pretty close to the minimum size, once linked to your actual application.

NullVoxPopuli, 2018-12-17:
@jedisct1 I don't know where to best ask this, but is libhypdrogen just one step in pulling out behavior from libsodium into separate packages?

closed 2019-11-21 by jedisct1