jedisct1/libhydrogen #20: Secure network communications

nxrighthere, 2018-04-18:
Hey Frank! First of all, thank you for this crypto library. I really like the Hydrogen, and at the moment I'm integrating it into my new networking layer.  Since I'm not familiar with this library, I want to make sure that I did everything right, so please, take a look at this scheme based on N variant key exchange:
1. A client connected to a server, and new public key generated for this client using `hydro_kx_keygen`
2. Public key sent to the client without any encryption to generate session keys using `hydro_kx_n_1`
3. Client send generated packet without any encryption to the server to compute session keys using `hydro_kx_n_2`
4. Now we can compute a secret key using `hydro_secretbox_keygen` and securely exchange encrypted messaged using `hydro_secretbox_encrypt` and `hydro_secretbox_decrypt`

jedisct1, 2018-04-18:
Hi!

This is even easier. You don't need to generate anything  in `4.`. `hydro_kx_n_2` already returns keys you can directly use with `hydro_secretbox_encrypt()` and `hydro_secretbox_decrypt()`.

jedisct1, 2018-04-18:
To clarify, `kp.rx` and `kp.tx` are filled with the session keys to use for encryption.

You can use the former to send data from the client to the server, and the later to send data from the server to the client. Or the other way round if you like.

nxrighthere, 2018-04-18:
Oh, thank you, Frank! I see now.

closed 2018-04-18 by nxrighthere