jedisct1/libhydrogen #22: Bindings

someburner, 2018-05-02:
Hi,

I'm hoping to use libhydrogen for an ESP8266-based project. I have successfully integrated most methods on the ESP, and wrote a C++ utility to sign/encrypt payloads for testing. But now we're working on integration and the lack of php or python bindings is unfortunate. I can probably write a python module, but neither I nor my colleague have experience writing a php extension from scratch. I thought [phosphine](https://github.com/absalomedia/phosphine) would be at least a start, but it appears to be just a skeleton with nothing actually implemented. 

Looking at your other repo [libsodium-php](https://github.com/jedisct1/libsodium-php) it looks like there might be a good amount of overlap in a libhydrogen port. So, is there any effort or plans for a port for libhydrogen?

someburner, 2018-05-03:
We got something going [here](https://github.com/phplang/hydro.git).

closed 2018-05-03 by someburner

jedisct1, 2018-05-03:
Oh, wow, this is neat!

Having bindings for all major languages is definitely needed. I started with Rust, but JS and PHP were obviously planned to be next.

PHP is by far the worst language to write bindings for, so I kept postponing it. But indeed, these should be pretty close to the libsodium ones, so quite a lot of code can just be reused.

Thanks for kicking this off!

someburner, 2018-05-03:
Np! thanks for libsodium + libhydrogen :+1:

someburner, 2018-05-05:
@jedisct1  - Made some progress on python, it's up here: https://github.com/someburner/pyhy

Only thing I had an issue with so far is cffi not liking `_hydro_attr_warn_unused_result_` and other method attributes. Not a big deal but it would be nice to source the file verbatim, so I'll try to see if there's any way around that that's clean.

someburner, 2018-05-09:
Added a demo with mqtt: https://github.com/someburner/pyhy/tree/master/demo

far enough to be added to the bindings wiki page?

jedisct1, 2018-05-09:
Great work, thank you!

jedisct1, 2018-05-11:
Awesome! Good luck, Fabian!

someburner, 2018-06-11:
Me again- I ended up re-writing a bunch of my services in go, so I made go bindings [here](https://github.com/someburner/libhydrogen-go). I'm still sorta new to go, so feedback welcome.

nxrighthere, 2018-12-01:
@jedisct1 C# wrapper is here https://github.com/nxrighthere/Hydrogen-CSharp ❤️