jedisct1/libhydrogen #34: Maximum length of message ?

arshidkv12, 2019-01-31:
What is the maximum length of message? 
Can I encrypt 100MB text file ?

jedisct1, 2019-01-31:
A message can be up to 2^64 bit long, so there is no practical limit.

The number of messages encrypted with the same key must remain below 2^64, which is rarely an issue either.

closed 2019-01-31 by jedisct1

arshidkv12, 2019-01-31:
Thank you... Can use this library for closed source project ? 
Like I will distribute only binary file.

jedisct1, 2019-01-31:
Still, for very large files, chunking is recommended. If only to save memory, and avoid processing a 100 MB ciphertext, only to discover later that it is invalid/corrupted.

This is fairly easy to do with libhydrogen, using message identifiers. Set `msg_id` to `0` for the first chunk, `1` for the second, etc. You can take a look at [encpipe](https://github.com/jedisct1/encpipe) for an example implementation.

And yes, you can totally use this in a closed source project, commercial or not, provided that a copy of the license is present somewhere.

arshidkv12, 2019-01-31:
Is there any way to reduce key side ? Like 32 characters lenth key.

arshidkv12, 2019-01-31:
Sorry for my basic questions.

jedisct1, 2019-01-31:
Exactly as explained above: derive a key from the short key/password using `hydro_pwhash_deterministic()` and use that key with `hydro_secretbox_encrypt/decrypt()`.

Or look at [encpipe.c](https://github.com/jedisct1/encpipe/blob/master/src/encpipe.c); it does exactly that.

arshidkv12, 2019-01-31:
Thank you very much