jedisct1/libhydrogen #36: Generating same secretbox key

daurnimator, 2019-02-02:
Calling `hydro_secretbox_keygen` in a new process gets the same result every time.

Delving into the code, this appears to be because `hydro_random_buf` doesn't call `hydro_random_check_initialized`

jedisct1, 2019-02-02:
Good catch, thanks!

But the application is expected to call `hydro_init()` before anything else.

closed 2019-02-02 by jedisct1

daurnimator, 2019-02-02:
> But the application is expected to call `hydro_init()` before anything else.

I didn't see that documented anywhere?

Why is `hydro_init` expected when `hydro_random_uniform` does the check itself?
(The main reason I could see it being useful is if you dropped permissions after startup so that retrieving outside random data would no longer be possible)

jedisct1, 2019-02-03:
First `hydro_init()` can be called before a `chroot()` call, so that `/dev/random` can be opened.

If you `chroot()` first, and call something that requires randomness later, this may fail.

More importantly, `hydro_init()` can do more than just initialize the PRG. For example, in libsodium, its counterpart `sodium_init()` selects the fastest implementations for the current CPU, and creates the canary for safe memory allocations.

The extra checks in `random*` are here for convenience, but they may eventually be turned into `abort()` calls.

PR #37 (2019-02-02): Check that PRNG has been initialised from hydro_random_buf
Fixes #36
commit 69a49a52 (2019-02-02): Check that PRNG has been initialised from hydro_random_buf

Fixes #36