jedisct1/libhydrogen #38: Why does hash have a minimum size?

daurnimator, 2019-02-02:
Why does `hydro_hash_BYTES_MIN` exist?
For a bloom filter I only want a 32 bit hash: why is there a 128 bit minimum on output hash size?

jedisct1, 2019-02-02:
Essentially to discourage people from using dangerously short tags for authentication. 128 bit is the bare minimum, leading to collisions after 2^64 messages.

Of course, there are totally valid use cases for shorter hash lengths, and a bloom filter is one of these.

In that case, you can truncate a 128 bit output to whatever size you need. By the way, bloom filters need more than one hash value for a given key. So you can take advantage of a 128 bit output, and split it into 4x32 values.

closed 2019-02-02 by jedisct1

daurnimator, 2019-02-04:
I've ended up with code all over the place that does `hash_final(max(16, wanted_length)):sub(1, wanted_length)`. I think the minimum should be removed.

daurnimator, 2019-02-04:
I've ended up with code all over the place that does `hash_final(max(16, wanted_length)):sub(1, wanted_length)`. As algorithm have adjustable size/parameters that can bring them down to needing < 16 bytes of hash data. I think the minimum should be removed.

jedisct1, 2019-02-05:
Bloom filters is not the most common use case for a hash function, and even then, you can leverage the 128 bits output for a slight speedup as described above. `crypto_hash` produces 128 bit values at a time since this is the rate's length, so a shorter output would be produced by truncating a 128 bit output anyway.

daurnimator, 2019-02-06:
> you can leverage the 128 bits output for a slight speedup as described above.

I know I *can*, but it's very awkward.

> `crypto_hash` produces 128 bit values at a time since this is the rate's length, so a shorter output would be produced by truncating a 128 bit output anyway.

As far as I can see, only the argument check needs to be removed. small outputs would already get handled correctly by [the leftover code](https://github.com/jedisct1/libhydrogen/blob/1514e4dd99772914bac8c91619b32d7c999cd806/impl/hash.h#L112)