jedisct1/libhydrogen #40: Porting to STM32 - what should hydro_random_init() do?

clintr-at-kornak, 2019-03-12:
Hello, I'm hoping to use LibHydrogen on an STM32L471 microcontroller.  My compiler reports the error "Unsupported Platform" in random.h.  It appears that I'll need to supply my own hydro_random_init() function.  Could you please tell me what it should do?  The microcontroller has a built-in hardware random number generator, if that helps.  Thanks.

jedisct1, 2019-03-12:
Hi!

Unfortunately, I don't have a board with that microcontroller, but if it has a built-in hardware RNG, this is going to be easy.

All you need is define a `hydro_random_init()` function for that platform, that fills `hydro_random_context.state`.

See how it's done on other platforms, particularly on ESP32.

clintr-at-kornak, 2019-03-12:
For anyone else who wants to use an STM32 with hardware RNG:
Copy the ESP32 version of hydro_random_init()
Use HAL_RNG_GenerateRandomNumber() instead of esp_random() to put random numbers in r.
Use HAL_Delay() instead of delay() - not sure there is a delay needed though.

closed 2019-03-12 by clintr-at-kornak