jedisct1/libhydrogen #42: error compiling with mingw

clintr-at-kornak, 2019-03-28:
Hello, I've been trying to compile with mingw (gcc version is 4.8.1) on Windows 7 64-bit.
"make lib" works, but for "make all" I get the error "undefined reference to `SystemFunction036@8'".  I get the same error if I try to compile libhydrogen.a into a program of my own.
Can you tell me how to correct this?
Thanks

clintr-at-kornak, 2019-03-28:
I've just tried updating to latest mingw (gcc version 8.2.0).  Problems are the same.  (I also had to put a blank line at the start of the makefile or make complained "make: cc: Command not found" - see https://stackoverflow.com/questions/36232323/mingw-make-cc-command-not-found).

jedisct1, 2019-03-28:
You should set the environment variable `CC` to the C compiler you want to use.

For example:

```
CC=x86_64-w64-mingw32-gcc
```

closed 2019-03-28 by jedisct1

clintr-at-kornak, 2019-03-28:
Hello, thank you for the solution to the mingw make error.  However, it does not solve my original problem, which is that I get the error "undefined reference to `SystemFunction036@8'".

jedisct1, 2019-03-29:
Have you tried with msys2?