jedisct1/libhydrogen #43: hydro_pwhash_create without a master key

spinda, 2019-04-05:
I understand the password-hashing API is designed to prevent misuse in the most common cases. But I'm using it to create a hash on a network client which is then transmitted—encrypted, in a secretbox

jedisct1, 2019-04-05:
No problem :) You can use `NULL` as the master key.

closed 2019-04-05 by jedisct1

spinda, 2019-04-06:
Great, thanks!

spinda, 2019-04-06:
Passing in `NULL` for the `master_key` parameter actually leads to a segmentation fault in `hydro_secretbox`, which (unlike `hydro_hash`) isn't set up to accept `NULL` key pointers. Passing in a pointer to an array of `NULL` bytes works for my purposes, but I could submit a patch to `hydro_secretbox` if this is intended to work.

jedisct1, 2019-04-06:
Being able to pass `NULL` as the master key was not the original intent. But as you asked, I saw that it would be fine with with `hydro_hash`, but indeed, we need a key for `secretbox`.

There's no point in trying to encrypt with a predicable key, so we should skip this if a `NULL` key is provided. This requires changes to all the current functions, but more importantly, I'm not sure that making encryption optional is a good idea.

This in fine for your use case, but in most applications, encryption is useful, even if this is with a static, hard-coded key.

spinda, 2019-04-07:
Alright! I can definitely understand making it harder to veer away from the known-safe path unless the developer is really sure what they're doing.