jedisct1/libhydrogen #53: Schnorr signatures

daurnimator, 2019-04-28:
For a project, I'm looking for an implementation of [Schnorr Signatures](https://en.wikipedia.org/wiki/Schnorr_signature) to store many signatures within a small constant size.

Is this something that would fit into libhydrogen?

jedisct1, 2019-04-28:
libhydrogen already implements Schnorr signatures (see the `hydro_sign*` API).

What it doesn't support is multi-parties signatures.

With BLS being currently standardized, I'm not convinced that there's still a point in using Schnorr signatures for multisig.

closed 2019-04-28 by jedisct1

daurnimator, 2019-04-28:
> What it doesn't support is multi-parties signatures.

What would it take to add it?

> With BLS being currently standardized, I'm not convinced that there's still a point in using Schnorr signatures for multisig.

AFAIK, BLS signatures are much more expensive to validate than plain schnorr signatures?

Either way, do you intend to add BLS signature support to libhydrogen?

jedisct1, 2019-04-28:
BLS signatures are computationally expensive, but besides being shorter, they have a huge advantage over Schnorr multi-signatures: they can be deterministic, i.e. they won't fail if the random number generator does. This is quite an interesting property for embedded systems.

That being said, I'm not planning to add either, as very few applications require multisigs, and libhydrogen should remain lightweight and constrained to what most application need.