jedisct1/libhydrogen #57: Unable to initialize on Android

bureto, 2019-05-10:
Hi, I am trying to build your library to run on android. The build process was successful, but the library wont initialize; I am building using android NDK. I made a [post ](https://stackoverflow.com/questions/55669064/native-library-not-found-on-android-build)on stackoverflow where there is more specific detail to my problem.

spinda, 2019-05-10:
Worth mentioning that this is not just Android but Unity on Android, so there's a whole other layer and language (C#) to pass through here.

A quick search brings up this [Unity-specific guide](https://docs.unity3d.com/Manual/AndroidNativePlugins.html) to native plugins on Android. Does the setup described there match your own?

bureto, 2019-05-11:
Hi spinda,
I went and did a thorough (triple) check of my Unity setup and all seems to be in line. I guess I should also make it clear that I am using [nxrighthere's wrapper](https://github.com/nxrighthere/Hydrogen-CSharp) to implement the library, which calls the native source (.so file) using the same naming as his [enet c# library](https://github.com/nxrighthere/ENet-CSharp). I have successfully been able to get that running on android, so it seems to be an issue with libhydrogen? If not, both native source are written in c, so I've kinda assumed it would be a very similar build and import process; maybe that is where I am wrong?

spinda, 2019-05-30:
Your logcat indicates that the library loader can't even find the `libhydrogen.so` file, though, so I suspect your problem isn't Hydrogen-specific.

> Copied the .so files from the `\libs` folder and placed them in their respective folders in Unity (i.e. `Hydrogen\Plugins\Android\arm64-v8a`).

Shouldn't that be `Assets\Plugins\Android\libs\arm64-v8a`?

bureto, 2019-05-31:
Although the Unity docs do say: 

> The "Plugins" folder is where you must put any native plugins, which you want to be accessible by your scripts. They will also be automatically included in your build. Take note that this folder may not be in any subfolder (it has to reside within the top-level Assets folder).

I believe (though I cannot say for certain) that this is wrong or outdated. It has always worked fine using separate sub-folder plugin directories, and I have tried merging all of the plugins in my project to one single root level folder in the past and it has broken all the imports. 

> I still suspect your problem isn't Hydrogen-specific.

I agree with you on this, I just ran some tests on an empty project and it worked! So something must be wonky with my project.. strange though that all other native libraries import fine. Thank you very much for you're patience and for keeping me on the right track, feel free to close this; I doubt anyone will run into the same issue as me but I will post the solution once I find it. :)

bureto, 2019-06-01:
I was finally able to pinpoint what was causing this.. it was a bug on Unity's part. When switching the project's target platform it seems to corrupt some of the files (in this case my hydrogen plugins). In hindsight I should have known since this wasn't the first time encountering such behavior. Strangely, it only seems to happen in this one project so something is definitely bonkers with it. 

Anyways, all is good now! Thank you for your time and for steering my attention towards Unity. I would probably have given up otherwise, since I was so convinced it had to do with hydrogen. I really appreciate it, thanks again.

closed 2019-06-01 by bureto