jedisct1/libhydrogen #61: hydro_hash disagrees with the public domain gimli hash when the total input length is an exact multiple of gimli_RATE

chrisnc, 2019-06-27:
It seems that the loop in hydro_hash_update checks whether gimli_RATE bytes have been fed in at the beginning of the loop, rather than the end, resulting in one fewer call to gimli overall if an integer multiple of gimli_RATE bytes are fed into hydro_hash_update, compared with the Gimli_hash function in the Gimli paper. https://gimli.cr.yp.to/gimli-20170627.pdf

chrisnc, 2019-06-27:
Attaching a log of what happens when I feed the string "test one two" into hydro_hash, compared with feeding the same data (including the prefix and suffix blocks), into Gimli_hash. Before and after each call to gimli I print the state.
[gimli.txt](https://github.com/jedisct1/libhydrogen/files/3333300/gimli.txt)
[hydrogen.txt](https://github.com/jedisct1/libhydrogen/files/3333301/hydrogen.txt)

closed 2019-06-27 by jedisct1

jedisct1, 2019-06-27:
Good catch, thank you!

commit a1876f9c (2019-06-27): hash: absorb the last full block

Fixes #61