jedisct1/libhydrogen #65: Problem encrypting some characters

naiteon, 2019-07-09:
Hi, I`ve noticed that it breaks when I try to encrypt some utf-8 chars like ã and é 

I tried encrypting a portuguese word "coração" (heart) and had a problem. 

Steps to reproduce erro:

1 . Encrypt: coração_é 
2. Decrypt the ciphertext generated
3 . Expected result: coração_é
4. Actual result: coraçã??-

jedisct1, 2019-07-09:
Hi,

None of the functions provided by the hydrogen API is parsing strings. Data are always handled as binary blobs.

Can you post a test case to illustrate the problem you are seeing?

naiteon, 2019-07-09:
Hi Frank, I have the following code, i  run it as ./encrypt text    i.e ./encrypt coração_é
It encrypts argv[1] then decodes it

#include <iostream>
#include <vector>
#include <sstream>
#include <string>
#include <hydrogen.h>

using namespace std;

#define CONTEXT "SOMECTXT"

int main(int argc, char* argv[]) {
    
    char *MESSAGE = argv[1];
    short MESSAGE_LEN = strlen(argv[1]);
    short CIPHERTEXT_LEN = hydro_secretbox_HEADERBYTES + MESSAGE_LEN;

    if (hydro_init() != 0) {
        abort();
    }
    
    cout << MESSAGE << " len = " << MESSAGE_LEN << endl;
    cout << " cyphertext len = " << CIPHERTEXT_LEN << endl;
    
    uint8_t key[hydro_secretbox_KEYBYTES];
    
    uint8_t ciphertext[CIPHERTEXT_LEN];
    
    hydro_secretbox_keygen(key);
    hydro_secretbox_encrypt(ciphertext, MESSAGE, MESSAGE_LEN, 0, CONTEXT, key);
    
    int count = sizeof(ciphertext) / sizeof(ciphertext[0]);
    
    cout << "ciphertext = " << ciphertext << endl;
    cout << "ciphertext len  = " << CIPHERTEXT_LEN << endl;

        
    char decrypted[CIPHERTEXT_LEN - hydro_secretbox_HEADERBYTES];

    if (hydro_secretbox_decrypt(decrypted, ciphertext, CIPHERTEXT_LEN, 0, CONTEXT, key) != 0) {
        cout << "some error\n";
    }

    cout << "decrypted text = " << decrypted << endl;

    
    return 0;
}

naiteon, 2019-07-09:
I`m testing on Mac OS X Mojave. You can also try with less characters ./encrypt coraçã  for example. What I find to be weird and may be related is that Mac OS is giving me strlen("coraçã") as 8 bytes. Seems like both ç and ã are considered 2 chars wide.

naiteon, 2019-07-09:
But even if I dont use those characters, this is what I get ./encrypt cor
results in cor?? and strlen("cor") as expected returned 3 size of bytes.

jedisct1, 2019-07-09:
Your terminal uses UTF-8 encoding. `ç` and `ã` are 1 character, but encoded as 2 bytes.

`strlen()` returns a string length, without the trailing `\0`.

But when you use `cout` to print a C string, that string needs a `\0` somewhere to know where it ends.

```c
char decrypted[CIPHERTEXT_LEN - hydro_secretbox_HEADERBYTES];
```

that should proably be

```c
char decrypted[CIPHERTEXT_LEN - hydro_secretbox_HEADERBYTES + 1];
decrypted[CIPHERTEXT_LEN - hydro_secretbox_HEADERBYTES] = 0;
```

Also in C and C++, value sizes are `size_t` not `short`.

Your issues don't have much to do with libhydrogen :)

naiteon, 2019-07-09:
Oh! Thank you so much I was getting nuts here, been trying to achieve it for 11 hours in a row! Thank you so much. I`ll close this issue

closed 2019-07-09 by naiteon