jedisct1/libhydrogen #69: Mbed compilator - error

rligocki, 2019-07-26:
I tried to compile libhydrogen on Mbed OS, but unfortunately I get this error:

```
[Error] @,: ARMC6 does not support ARM architecture v7 targets
[Warning] @,: Compiler version mismatch: Could not detect version; expected version >= 6.10 and < 7.0
Building project mbed-os-empty (NUCLEO_F429ZI, ARMC6)
Scan: mbed-os-empty
Link: mbed-os-empty
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6200E: Symbol hydro_sign_verify multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_kk_3 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_hash_init multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_secretbox_probe_create multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol __ARM_use_no_argv multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/tests/tests.o).
Error: L6200E: Symbol hydro_kdf_derive_from_key multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_n_1 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kdf_keygen multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pwhash_verify multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pwhash_deterministic multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_secretbox_encrypt multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_sign_keygen multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pad multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_xx_3 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_hash_final multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_n_2 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_sign_final_create multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_secretbox_keygen multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_secretbox_decrypt multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_xx_1 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_random_buf_deterministic multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_kk_2 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_random_reseed multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_secretbox_probe_verify multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pwhash_keygen multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_init multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_random_ratchet multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pwhash_create multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_sign_keygen_deterministic multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_xx_4 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_xx_2 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_hex2bin multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_memzero multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_hash_hash multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_bin2hex multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_keygen multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_sign_init multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_random_buf multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol main multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/tests/tests.o).
Error: L6200E: Symbol hydro_unpad multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_increment multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_random_uniform multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_sign_update multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_hash_update multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pwhash_upgrade multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_kk_1 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_equal multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_random_u32 multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_sign_final_verify multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_sign_create multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pwhash_derive_static_key multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_compare multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_pwhash_reencrypt multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_kx_keygen_deterministic multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
[... cut ...]
Error: L6200E: Symbol hydro_kx_keygen_deterministic multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Error: L6200E: Symbol hydro_hash_keygen multiply defined (by BUILD/NUCLEO_F429ZI/ARMC6/main.o and BUILD/NUCLEO_F429ZI/ARMC6/libhydrogen/hydrogen.o).
Not enough information to list the image map.
Finished: 1 information, 1 warning and 55 error messages.
```

Does anyone know what to do? 

This is my code:

```
#include` "mbed.h"
#include "./../x86_example/hydrogen.cpp"

// main() runs in its own thread in the OS
int main()
{
    while (true) {
        //hydro_init();
    }
}
```

jedisct1, 2019-07-26:
What is `x86_example/hydrogen.cpp`?

rligocki, 2019-07-27:
It is the second example for x86 architecture. It is at the same workspace. Not Mbed project. I tried also 
`#include "libhydrogen/hydrogen.cpp"`

Could it be a problem that in one workspace I have more than one directory with libhydrogen?

rligocki, 2019-07-27:
Now I tried:

`#include "hydrogen.c" `

and still, have the same problem.

jedisct1, 2019-07-27:
Unfortunately I don't know what examples and development environment you are talking about.
But if `hydrogen.c` is already automatically compiled because it is in the same workspace, you should `#include "hydrogen.h"`, not the `.c` file.

rligocki, 2019-07-28:
Thanks !!! #include "hydrogne.h" works !! I used code from Visual Studio, where "#include "hydrogen.c"" worked.

closed 2019-07-29 by jedisct1