jedisct1/libhydrogen #71: tests: failed to build the tests on Windows in Release mode

myd7349, 2019-08-10:
Currently, the test suite uses the `assert` macro to report test results. However, this macro is usually expand to `(void)0` in Release mode. As a result, there will be some warnings:
> c:\users\vcpkg\desktop\libhydrogen\tests\tests.c(125): warning C4101: 'hexf': unreferenced local variable
c:\users\vcpkg\desktop\libhydrogen\tests\tests.c(123): warning C4189: 'b': local variable is initialized but not referenced
c:\users\vcpkg\desktop\libhydrogen\tests\tests.c(121): warning C4101: 'y': unreferenced local variable
c:\users\vcpkg\desktop\libhydrogen\tests\tests.c(169): warning C4101: 'm2': unreferenced local variable
c:\users\vcpkg\desktop\libhydrogen\tests\tests.c(371): warning C4101: 'stored': unreferenced local variable
c:\users\vcpkg\desktop\libhydrogen\tests\tests.c(373): warning C4101: 'static_key': unreferenced local variable
when building in Release mode with MSVC.

Unfortunately, a [```/WX```](https://docs.microsoft.com/en-us/cpp/build/reference/wx-treat-linker-warnings-as-errors?view=vs-2019) option is also specified In `CMakeLists.txt` when using MSVC:
https://github.com/jedisct1/libhydrogen/blob/68e3c591db860e806131d8c1d5ef842674fc87aa/CMakeLists.txt#L81
This options makes MSVC treat all warnings as errors. And the build will fail.

##### How to reproduce?
```
mkdir build
cmake .. -G "Visual Studio 14 2015"
cmake --build . --config Release
```

I think we may fix this in these ways:
1. We define a `assert` like macro ourselves so that it also works in Release mode. It seems a good idea since we can also run the tests in Release mode.
2. Remove the `/WX` in `CMakeLists.txt`. This fix is short, and that is what I did in https://github.com/microsoft/vcpkg/pull/7436 .

jedisct1, 2019-08-10:
Thanks!

`NDEBUG` is now unconditionally undefined, in case alternative build systems such as `cmake` define it.

`/WX` has been removed as well.

closed 2019-08-10 by jedisct1

spinda, 2019-08-10:
I'm not so sure this is the right fix. What's the value of building the tests in Release mode, if we allow `assert` to turn into `(void)0` in that mode? Then the tests won't actually test anything.

jedisct1, 2019-08-10:
`NDEBUG` being undefined should be enough to have the `cmake` version behave like the Makefile.

commit b3a2a302 (2019-08-10): Remove /WX from the CMakefile, and always undefine NDEBUG

Also, we currently don't need <assert.h> in the library

Fixes #71