jedisct1/libhydrogen #73: Private key decryption only working after printf

solanav, 2019-08-10:
Currently I'm using the XX key exchange and after that I encrypt a message on a server and decrypt it on the client. The decryption only works if I print the encrypted data right before calling hydro_secretbox_decrypt. I have to print at least 1 byte of the encrypted buffer for it to work.

Encryption:
```
#define MAX_UDP 512
uint8_t all_data[MAX_UDP - hydro_secretbox_HEADERBYTES];
uint8_t encrypted_data[MAX_UDP];

...

hydro_secretbox_encrypt(encrypted_data, all_data, MAX_UDP - hydro_secretbox_HEADERBYTES, 0, "debug", key);
memcpy(datagram, encrypted_data, MAX_UDP);
```

Decryption:
```
uint8_t decrypted_data[MAX_UDP - hydro_secretbox_HEADERBYTES];
uint8_t key[hydro_secretbox_KEYBYTES];

sem_wait(sem);
memcpy(key, sd->peers.kp[peer_index].rx, hydro_secretbox_KEYBYTES);
sem_post(sem);
if (hydro_secretbox_decrypt(decrypted_data, data,
						MAX_UDP, 0,
						"debug", key) == -1)
{
	DEBUG_PRINT(P_ERROR "Failed to decrypt the message\n");
	return ERROR;
}
```

I have also manually checked that the encrypted message is not tampered with (by printing it in both ends of the connection) and that the key is correctly generated before sending anything.

jedisct1, 2019-08-10:
You are likely doing out of bound memory accesses.

It may be somewhere else, but there's at least one thing that doesn't look right in the code you posted: the context must be 8 bytes (`hydro_secretbox_CONTEXTBYTES`). The `"debug"` string is too short.

solanav, 2019-08-10:
Well that was disappointingly easy to fix. I just changed "debug" to "examples" and it works perfectly. Thank you for your help and for writing this amazing library!

jedisct1, 2019-08-10:
Or you can pad with extra zeros.

closed 2019-08-10 by jedisct1