jedisct1/libhydrogen #78: Nuttx support

rligocki, 2019-09-30:
Hi, 

I found way how to generate random value on Nuttx real time OS, and now I want to add support for nuttx in libhydrogen. To add support I need to define function hydro_random_init(void) for NuttX. Which value should I enter into defined() or #ifdef condition under which new hydro_random_init function will be defined?

Thanks

rligocki, 2019-10-02:
Finally, I found an environment variable for PX4. It's #elif  defined(__PX4_NUTTX). I also found way how to generate random seed for libhydrogen. Is it possible to add a new branch and then create a pull request? 

I already tried to create a new branch and push a new commit, but my permissions were denied.

jedisct1, 2019-10-02:
You need to fork the project first :)

closed 2019-12-27 by jedisct1