jedisct1/libhydrogen #79: libhydrogen doesn't compile on esp8266 - esp_system.h missing

ulno, 2019-10-01:
Just tried to add libhydrogen to one of my esp8266 projects.

It fails compiling random.h in line 76 as esp_system.h doesn't exist.

AFAIK, this file exists only on ESP32 platforms.
When I comment the #include line, the compilation works, so I assume there is another ``#ifdef ESP32`` needed?

jedisct1, 2019-10-02:
Hi,

That's very likely.

But even if compilation works, it would be reasonable to take the time to verify that the RNG works as expected.

Unfortunately, the only microcontroller I have is an Arduino, so your help is badly needed :)

ulno, 2019-10-08:
just providing an empty esp_system.h solves the problem for me now. I will report back with some results soon using libhydrogen on esp8266 soon.

PR #107 (2021-05-23): Do not include esp_system.h on nodemcu
We need to include esp_system.h on ESP32 only

Fixes #79
closed 2019-12-27 by jedisct1