jedisct1/libhydrogen #81: Questions regarding Gimli as a Secure hash Function

DonaldTsang, 2019-10-18:
Is Gimli as secure as SHA-256 or SHA-512? What about SHA3 or SHAKE?
Because SHA3's security is equals to four times the size of the capacity of the permutation...
(I mean for SHAKE it is two times the capacity of the permutation but Gimli can be used as a hash)
BUT Gimli's 256-bit hash has the capacity of 256, which means it has only 64-bit security to SHA3?
Unless it is meant to be a XOF-only, in that case it should be 128-bits normally?

jedisct1, 2019-10-18:
Gimli is not meant to be used as a hash function. With the default parameters, it would not be collision resistant.
It is indeed meant to be used as a XOF. The construction implemented in libhydrogen is an instantiation of kMAC with Gimli. It has a 128 bit security level if the goal is to find *any* collision. The practical security is much higher.

closed 2019-10-18 by jedisct1

chrisnc, 2019-10-28:
> Gimli is not meant to be used as a hash function. With the default parameters, it would not be collision resistant.

@jedisct1, can you give some more details about this? If I'm reading it correctly, the spec paper from earlier this year (https://csrc.nist.gov/CSRC/media/Projects/Lightweight-Cryptography/documents/round-1/spec-doc/gimli-spec.pdf) claims that gimli-hash (as defined with its default 256-bit output) has 128-bit security and is intended to be collision resistant. What am I missing here?

jedisct1, 2019-10-29:
Gimli-hash, that is the Gimli permutation used in a spongy construction, is collision resistant.

OP's question isn't very clear, but I interpret it as using the Gimli permutation function as a compression function.

DonaldTsang, 2019-10-29:
@chrisnc thanks for the clarification, I did read the paper which raised a few questions.
@jedisct1 How collision resistant and pre-image resistant is Gimli-hash anyway?