jedisct1/libhydrogen #82: update hash to match new gimli spec

chrisnc, 2019-10-29:
The Gimli hash candidate in the NIST Lightweight Cryptography Project pads the state differently than the reference code in the original 2017 paper does. It XORs 0x01 into the byte following the last incomplete block, and XORs 0x01 into the last byte of the state (not the 16th byte).

Is there a plan to update libhydrogen to match the new specification?

jedisct1, 2019-10-29:
Hi Chris,

libhydrogen uses the KMAC construction. It doesn’t use the Gimli mode as its specification was not finalized, probably still isn’t, and doesn’t bring much over KMAC.

closed 2019-10-29 by jedisct1

jedisct1, 2019-10-29:
Gimli hash is not exposed in the public API. Unless other libraries use it within KMAC, which is very unlikely, libhydrogen's hash API is only compatible with itself.
Changing the internal hash function is not going to improve this. It would only make it incompatible with itself. I don't think it's worth it.