jedisct1/libhydrogen #85: key exchange with pre-shared key works when psk do not match

llsoft2000, 2019-11-21:
Hi,

I did some tests with the key exchange (N variant) focusing on the pre-shared key feature.
In my tests I used different combinations of pre-shared keys for hydro_kx_n_1 and hydro_kx_n_2.
I expected hydro_kx_n_2 to fail when a different pre-shared key was used than the one for hydro_kx_n_1.
I tested the following combinations:

hydro_kx_n_1 | hydro_kx_n_2 | test result | expectation | comment
------------------- | ------------------- | -------------| ---------------- | ------------
NULL | NULL | passed | pass | 
psk | psk | passed | pass | covered by testsuite
psk | NULL | passed | failure | 
NULL | psk | passed | failure | 
psk | psk + 1 | passed | failure | 

I would like to know if I do something wrong using kx_n this way?
How can I reliably deduce from calling hydro_kx_n_2 if both PSK were the same?
Thank you very much.

Best regards,
llsoft2000

```c
static void
test_kx_n_NULL_psk(void)
{
    hydro_kx_keypair         server_static_kp;
    uint8_t                  psk[hydro_kx_PSKBYTES];
    uint8_t                  packet1[hydro_kx_N_PACKET1BYTES];
    hydro_kx_session_keypair kp_client;
    hydro_kx_session_keypair kp_server;
    int i1, i2;

    hydro_memzero(&kp_client, sizeof kp_client);
    hydro_memzero(&kp_server, sizeof kp_server);

    hydro_kx_keygen(&server_static_kp);
    hydro_random_buf(psk, sizeof psk);

    i1 = hydro_kx_n_1(&kp_client, packet1, NULL, server_static_kp.pk);
    i2 = hydro_kx_n_2(&kp_server, packet1, psk, &server_static_kp);

    assert(0 == i1);
    assert(0 != i2);

    assert(!hydro_equal(kp_client.tx, kp_server.rx, hydro_kx_SESSIONKEYBYTES));
    assert(!hydro_equal(kp_client.rx, kp_server.tx, hydro_kx_SESSIONKEYBYTES));
}

static void
test_kx_n_psk_increment(void)
{   
    hydro_kx_keypair         server_static_kp;
    uint8_t                  psk[hydro_kx_PSKBYTES];
    uint8_t                  packet1[hydro_kx_N_PACKET1BYTES];
    hydro_kx_session_keypair kp_client;
    hydro_kx_session_keypair kp_server;
    int i1, i2;
    
    hydro_memzero(&kp_client, sizeof kp_client);
    hydro_memzero(&kp_server, sizeof kp_server);
    
    hydro_kx_keygen(&server_static_kp);
    hydro_random_buf(psk, sizeof psk);
    
    i1 = hydro_kx_n_1(&kp_client, packet1, psk, server_static_kp.pk);
    hydro_increment(psk, sizeof psk);
    i2 = hydro_kx_n_2(&kp_server, packet1, psk, &server_static_kp);
    
    assert(0 == i1);
    assert(0 != i2);
    
    assert(!hydro_equal(kp_client.tx, kp_server.rx, hydro_kx_SESSIONKEYBYTES));
    assert(!hydro_equal(kp_client.rx, kp_server.tx, hydro_kx_SESSIONKEYBYTES));
}
```

jedisct1, 2019-11-21:
Checking if a PSK is the same both sides cannot be done with a single round trip. That would require an extra round trip.

However, if the PSKs are different, the shared keys are not going to match. Anything authenticated with one of these keys (such as with `hydro_secretbox`, or `hydro_secretbox_probe`) will immediately fail.

closed 2019-11-21 by jedisct1

llsoft2000, 2019-11-22:
Thank you for the reply.
You write that the shared keys are not going to match if the PSK are different.
I assume by 'shared keys' you mean the session keys (tx, rx) generated with hydro_kx_n_1 and hydro_kx_n_2.
But when I use these keys (tx, rx) with hydro_secretbox, nothing fails even if the PSK do  not match.

The following test case fails. I think it reflects what you explained in your answer.

```c
static void
test_kx_n_psk_fail(void)
{
    uint8_t m[25];
    uint8_t c[hydro_secretbox_HEADERBYTES + 25];
    uint8_t probe[hydro_secretbox_PROBEBYTES];
    uint8_t packet1[hydro_kx_N_PACKET1BYTES];
    uint8_t psk[hydro_kx_PSKBYTES];
    hydro_kx_keypair server_static_kp;

    // server
    hydro_kx_keygen(&server_static_kp);

    // PSK
    hydro_random_buf(psk, sizeof psk);

    { // client
      uint8_t dk[hydro_random_SEEDBYTES];
      hydro_kx_session_keypair kp_client;

      memset(dk, 0, sizeof dk);
      hydro_random_buf_deterministic(m, sizeof m, dk);

      hydro_kx_n_1(&kp_client, packet1, psk, server_static_kp.pk);
      hydro_secretbox_encrypt(c, m, sizeof m, 0, ctx, kp_client.tx);
      hydro_secretbox_probe_create(probe, c, sizeof c, ctx, kp_client.tx);
    }

    { // server
      hydro_kx_session_keypair kp_server;
      uint8_t m2[25];

      // same PSK on client and server
      memset(m2, 0, sizeof m2);
      hydro_kx_n_2(&kp_server, packet1, psk, &server_static_kp);
      assert(hydro_secretbox_probe_verify(probe, c, sizeof c, ctx, kp_server.rx) == 0);
      assert(hydro_secretbox_decrypt(m2, c, sizeof c, 0, ctx, kp_server.rx) == 0);
      assert(hydro_equal(m, m2, sizeof m));

      // different PSK on client and server
      hydro_increment(psk, sizeof psk);
      memset(m2, 0, sizeof m2);
      hydro_kx_n_2(&kp_server, packet1, psk, &server_static_kp);
      assert(hydro_secretbox_probe_verify(probe, c, sizeof c, ctx, kp_server.rx) == -1);
      assert(hydro_secretbox_decrypt(m2, c, sizeof c, 0, ctx, kp_server.rx) == -1);
      assert(!hydro_equal(m, m2, sizeof m));
    }
}
```