jedisct1/libhydrogen #90: Noise KK between multiple peers

Crest, 2020-03-28:
I need a one roundtrip handshake like Noise KK for a protocol on top of UDP and have to use a single UDP socket for a lot of reasons. The problem is that Noise KK requires the responder to implicitly know the initiators identity. I worked around this by using Noise N to send the encrypted long term public key along. This works, but requires two public keys (one for N and one for KK) per node which obviously doubles the size of the key material. Is it save to reuse the same key pair for both key exchange variants in this construction or should two key pairs be derived from a single seed?

jedisct1, 2020-03-28:
Hi Crest,

Using the same public key for both multiple Noise variants should be totally fine.

Crest, 2020-03-29:
Thanks for answering my question. Using a single longterm keypair for the different Noise variants (N and KK) simplifies my code and the configuration expected of its eventual users.

closed 2020-03-29 by Crest