jedisct1/libhydrogen #91: nRF52 without softdevice

wombatsoftwareguild, 2020-04-15:
There are plenty of uses for libhydrogen on an nRF52 when the softdevice is not present or not enabled. There is a separate nRF5 SDK set of API calls for the random number generator when the softdevice APIs are not availaible. (I'm not sure if one can use these APIs when the softdevice is enabled -- it's tough to sort through the Nordic documentation.)

Request modifying impl/random.h to use the alternate random API if SOFTDEVICE_PRESENT or sd_softdevice_is_enabled() indicates the softdevice APIs are not available.

wombatsoftwareguild, 2020-04-15:
Actually `nrf_sdh_is_enabled()` might be a better choice for checking which API to use.

jedisct1, 2020-04-15:
Go for it :)

wombatsoftwareguild, 2020-04-20:
Just tried to create a tracking branch on git@github.com:jedisct1/libhydrogen.git but got permission denied. Odd, because at least briefly I though GitHub listed me as a collaborator.

Do you prefer a PR or emailed diffs?

 =====

closed 2025-01-09 by github-actions[bot]