jedisct1/libhydrogen #92: Decrease required cmake version

ohell, 2020-05-16:
Right now CMakeLists.txt requires cmake version 3.12+.  This is higher than the version available on any Ubuntu LTS distribution. Just decreasing this version will make building a lot easier without impacting anything.
cmake 3.1 will be sufficient for our purposes, but it might be possible to go even lower.

jedisct1, 2020-05-16:
The current Ubuntu LTS seems to be shipping with version 3.16.3.

But would changing `cmake_minimum_required` to version `3.1` be enough to make it compatible with `cmake` 3.1 or are more changes needed?

ohell, 2020-05-16:
Sorry, I should have been clearer - Ubuntu 18.04 is the latest LTS that is supported by docker, and that only has 3.1. So not possible to containerize projects that use hydrogen, unless sources are included directly in the project. But then tests are not run  when building.

I just checked, cmake 3.1 does not support list:TRANSFORM verb. So there will be changes. I will try to get it working, and submit a pull request if I succeed.

Thanks.

closed 2020-05-17 by ohell