jedisct1/libhydrogen #94: Using libhydrogen in secure bootloader

Laczen, 2020-07-04:
Hi, I would like to use libhydrogen in a secure bootloader for a microcontroller (nrf81522, nrf52832 or similar). 

The bootloader has 2 slots, one for unencrypted firmware (slot 0) and one for encrypted (slot 1) firmware. The encrypted firmware is updated. A firmware upgrade unencrypts the new firmware to the unencrypted slot and at the same time encrypts the old firmware to the encrypt slot. 

As these devices only have limited ram the encrypt/decrypt is done one flash page at a time: copy a flash page from slot 1 to a temporary location, erase the flash page in slot 1, encrypt a flash page from slot 0 to slot 1 (using key 0), decrypt from temporary location to slot 0 (using key 1), and continue with the next page.

Is such a scheme possible with libhydrogen?

jedisct1, 2020-07-05:
Hi!

A simple way to achieve this would be to split the input into pages, and, for each page, call `hydro_secretbox_encrypt()` with the page number as the `msg_id` parameter.

But encryption has a 36 bytes overhead, which can add up as pages are encrypted individually.

What is the page size?

If that turns out to be a problem, it should be fairly easy to modify the code to use a bounce buffer instead of copying data linearly.

Laczen, 2020-07-05:
@jedisct1, thanks for the quick reply. 

The page size is at least 2kB, the overhead is not really a problem (size aspect), the problem is that it is  no longer possible to use the sector approach. 

I will have a look at modifying the code to use a bounce buffer.

Another problem I'm having is that it should be possible to continue decryption after power fail, in this case there willl be some sectors of slot 0 containing unencrypted new firmware and other sectors containing unencrypted old firmware and vice versa for the encrypted firmwares in slot 1.

closed 2020-07-06 by Laczen