jedisct1/libhydrogen #102: Last step: register libhydrogen on TIS CI

guillaumemillot, 2020-10-07:
Thanks for merging https://github.com/jedisct1/libhydrogen/pull/99 that contains an initial TIS CI configuration!

For TIS CI to start analyzing libhydrogen, you need to do one more thing: register the project on TIS CI from your side.

Here are the steps:
1. Go to https://ci.trust-in-soft.com/ and sign up,
2. Once on TIS CI, click on **Add Project** in the top navigation menu,
3. Select libhydrogen from the project list,
4. Click on **Start verifying**.

From this point on, the badge on the README will become green after the first analysis and all pushes to the repository will trigger a new analysis.

Thanks!

jedisct1, 2020-10-07:
Done, thank you!

jedisct1, 2020-10-07:
Is it possible to verify multiple branches?

I may have missed a setting somewhere, but I didn't see how to add branches besides the main one when adding a project. Sorry for the stupid question :)

closed 2020-10-07 by jedisct1

guillaumemillot, 2020-10-07:
You're actually not the only one asking this question, which means we should really clarify what happens when a project is added :)

Yes, you can analyze multiple branches. After you've added a project, "just" push the TIS CI configuration, including the `tis.config` file, to the branch(es) that you'd like to analyze. Note: always place the `tis.config` file at the root of the repository.

What's currently not clear I think is: the branch that is selected in Add project is only used to determine the first commit (the HEAD of that branch) that will be analyzed.

Thanks by the way for adding the project! If you're interested, we're preparing a similar configuration for libsodium. If you don't mind, I'll have a few questions (the project is bigger so it's a bit more complex to set up).