jedisct1/libhydrogen #106: Gimli out of running in LWC competition

woefulwabbit, 2021-03-30:
https://csrc.nist.gov/News/2021/lightweight-crypto-finalists-announced

Gimli did not make it to the finals of the LWC competition. Are there any plans to switch to one of the other finalists, or perhaps wait for the overall winner to be announced?

mko-io, 2021-03-30:
What a shame, Gimli is thing of beauty. Even Intel has found it really good

A recent Intel paper “Gimli encryption in 715.9 psec” by Intel

pepa65, 2021-03-30:
I wonder about the rationale of it not making round 3, it might be immaterial to the encryption security itself..?

jedisct1, 2021-03-30:
There's nothing wrong with Gimli, and even though it quickly became the center of attention, no practical issues were found.

But they had to limit the number of finalists. Xoodoo is in the same ballpark, is a little bit more conservative, and has more specified mode. So, they picked Xoodyak, and it is still good to see that they didn't pick only primitives that are efficient in hardware.

Libhydrogen may switch to Xoodoo, but there's no rush. No strong justification for breaking existing applications using libhydrogen.

Xoodyak doesn't include NMRAE modes, so this is not something that can be used as-is in libhydrogen anyway.

closed 2021-03-30 by jedisct1

mko-io, 2021-03-30:
@jedisct1 What do you think about the Gimli Hash and Gimli AEAD implementation from the later published paper?

jedisct1, 2021-03-30:
Nothing special. This is the regular duplex mode. There are more efficient modes such as Beetle used by other candidates.