jedisct1/libhydrogen #114: Linking issue on Android

AshishDurgapal, 2021-07-29:
Hi,

I am not able to compile the libhydrogen source code in android as it is throwing error while liking with test sample (tests.c).

Please check the given below linking errors.

**_/home/Compiler/android_aarch64_clang_toolchain3.5/bin/../lib/gcc/aarch64-linux-android/4.9/../../../../aarch64-linux-android/bin/ld: /tmp/hydrogen-d0e9ef.o: unrecognized relocation (0x20c) in section `.text'
/home/Compiler/android_aarch64_clang_toolchain3.5/bin/../lib/gcc/aarch64-linux-android/4.9/../../../../aarch64-linux-android/bin/ld: final link failed: Bad value_**

How to Reproduce : 
I have added a entry of android toolchain in make file and then just run make.

Please let me know what causing this issue.

jedisct1, 2021-07-29:
Feels to me that some of the tools are from the Android tool chains, but they get mixed by other tools with the same name, also in your PATH, but for a different architecture.

Given the simple structure of that project, it is probably easier to just ignore the Makefile and import the files directly in your project.

AshishDurgapal, 2021-07-29:
Hi @jedisct1 ,

I am getting the same error when I include the file directly in my project.

When I link the library made from inclusion of hydrogenlib source code then it is throwing same error while it is working fine with Windows, Linux and MacOS.

Is there any special flag that I have to include ?

AshishDurgapal, 2021-07-30:
Is this source is compatible with android platform or we have to port the code for android platform.?

jedisct1, 2021-07-30:
Can you compile a simple `hello world` or `add()` function also written in C?

AshishDurgapal, 2021-07-30:
Yes, I am able to compile and run the C sample using same tool chain and it is working fine.

closed 2025-01-09 by github-actions[bot]