jedisct1/libhydrogen #132: How to send a message using KK_variant Key exchange?

KamiK2K, 2022-08-03:
Hi,

In my case server and the client know their peer's public keys and I am using KK-Variant Key exchange algorithm to perform an Asymmetric encryption. I need to encrypt a message using a remote device's public key and send it to the remote device, and vice versa. 

How does this message transfer work in the KK variant, I don't understand how to use session_kp.tx and session_kp.rx to send and receive the data to and from a remote device respectively.

Has anybody done this before that can help me please?

Thanks

Crest, 2022-08-03:
Have you already found the wiki article on the [KK-variant](https://github.com/jedisct1/libhydrogen/wiki/KK-variant)?

KamiK2K, 2022-08-04:
Thank you for your answer, 
Yes I read most of the wiki documentation for this library along with [https://github.com/jedisct1/libhydrogen/wiki/KK-variant](url) and [https://github.com/jedisct1/libhydrogen/wiki/Secret-key-encryption](url).
Based on the Author's comment here [https://github.com/jedisct1/libhydrogen/issues/16](url); I believe that we can use this same function for asymmetric encryption when we have computed the keys from on of the  [key exchange mechanisms](https://github.com/jedisct1/libhydrogen/wiki/Key-exchange).
Is this correct?

I am little bit confused on how to use the KK-Variant and send a encrypted message to a remote device and vice versa.

Any help is truly appreciated,
Kami

jedisct1, 2022-08-04:
After the key exchange (using `KK` or any other mechanism), you end up with two keys.

Use the first one as a `secretbox` key to send data in one direction (for example for messages sent by the initiator to the receiver), and the second one also with `secretbox` in the other direction (for example for messages sent by the receiver to the initiator).

KamiK2K, 2022-08-04:
Thank you for your response @jedisct1 
I think I got your point, 
thank you for your support and help.

KamiK2K, 2022-08-05:
Hi Again,

I am using KK_Variant and calling hydro_kx_kk_1( ) function, which initializes the client_st variable, writes to packet1;

Then when I call the second function hydro_kx_kk_2( ) and pass packet1 along with the peer's public ley and local key pair, it doesn't load packet2 and does not generate the session key pair for the server, 

when I check the following conditional statement:

if (hydro_kx_kk_2(&session_kp, packet2, packet1,
    client_static_kp.pk, &server_static_kp) != 0) {
  // abort
}
it doesn't enter inside the if statement, which means the function operates properly as intended, but I don't get the outputs that I need from this function!

Similar thing happens for my hydro_kx_kk_3 () function call.

Can somebody let me know what am I missing in using this function?

Any help is much appreciated :)
Kami

KamiK2K, 2022-08-05:
I think I got my answer on Authors comment on [isuue8](https://github.com/jedisct1/libhydrogen/issues/8) where he says:

kx_keypair() "clamps" the secret scalar. The 3 low bits are cleared to make it a multiple of the cofactor, and prevent small subgroup attacks. Since a single party is required for signing, sign_keypair() doesn't perform this clamping operation. Using a key exchange key pair for signatures will work, but not the other way round.

Thank you :)

closed 2022-08-18 by KamiK2K