jedisct1/libhydrogen #133: Leaked public keys of KK_variant?

KamiK2K, 2022-08-17:
Hi, 

I am trying to use the KK_variant of the key exchange mechanism, where public keys of parties (generated using hydro_kx_keygen function) is exchanged in the clear. Does this endanger the security of the tx and ex session key pairs?

This is considering the fact that I have authenticated both sides with signature verification. Detailed steps is the attached picture (both sides know each other public keys for signature verification). 

[image]

Crest, 2022-08-18:
Is the connection between device and PC in step 3 trusted in your design?

KamiK2K, 2022-08-18:
No its not trusted in that step, its a Bluetooth or USB connection,

In other words the ephemeral public key of the device generated at step 3 can be sniffed by other devices,

But I regenerate those keys using hydro_kx_keygen() function call to provide forward secrecy for each session,

each session usually lasts few hours,

so in this case, my main question is if the public keys of both sides generated using hydro_kx_keygen() function for each session gets received by other devices, does that impose a threat for security of that session? Can the attacker somehow recover the session keys? and decrypt the exchanged messages?

jedisct1, 2022-08-18:
This is perfectly fine. Public keys can be leaked (hence "public"); it doesn't allow an attacker to recover the session key without the corresponding secret key.

closed 2022-11-08 by jedisct1