jedisct1/libhydrogen #135: Libhydrogen API hydro_kx_kk_2 getting failed in AIX

AshishDurgapal, 2022-09-01:
We have static keys sender private key, sender public key and recipient public key but when we send the packet1 from client to server, then server failed to decrypt the packet using API( hydro_kx_kk_2 ) in AIX platform.

However, same keys are working fine in linux, windows, MAC and android, so could you please let me know why we are not able to decrypt the packet over AIX only.

one more observation, We have generated the static key pairs using API hydro_kx_keygen but keys generated in AIX is not working with linux and vice-a-versa.

jedisct1, 2022-11-08:
Maybe an endianness issue.

Do you have a code snippet that reproduces this?

AshishDurgapal, 2022-11-14:
Correct. 
The issue was due to endianness and we have fixed the issue by using flag .
Thanks @jedisct1 for your time.

closed 2025-02-11 by github-actions[bot]