jedisct1/libhydrogen #136: Error C4146 in Windows

yuppox, 2022-10-23:
I'm seeing 4 errors in core.h on Windows. All of them are:
```
Error C4146 unary minus operator applied to unsigned type, result still unsigned
```

Here is the first instance

```
    for (i = 0; i < blocksize; i++) {
        barrier_mask = (unsigned char) (((i ^ xpadlen) - 1U) >> ((sizeof(size_t) - 1U) * CHAR_BIT));
        tail[-i]     = (tail[-i] & mask) | (0x80 & barrier_mask);
        mask |= barrier_mask;
    }
```

The second instance is here:

```
    for (i = 0U; i < blocksize; i++) {
        c          = tail[-i];
        is_barrier = (((acc - 1U) & (pad_len - 1U) & ((c ^ 0x80) - 1U)) >> 8) & 1U;
        acc |= c;
        pad_len |= (i & -is_barrier);
        valid |= (unsigned char) is_barrier;
    }
```

The problems are `tail[-i]` and `-is_barrier`. They are all `size_t` which is unsigned.

Why are those negative? I especially don't understand `tail[-i]`. It's counting from 0 to some size. Why would you want that to be negative when accessing an array?

jedisct1, 2022-10-23:
`p[x]` is equivalent to `*(p + x)`. So a negative value for `x` [is not supposed to be an error](https://stackoverflow.com/questions/36347748/error-c4146-unary-minus-operator-applied-to-unsigned-type-result-still-unsigne).

Anyway, the notation was changed to `*(p + x)` in 63260875a1b66fd87d210d126406e88fde483706 which also makes the consistent with libsodium.

closed 2022-10-23 by jedisct1

yuppox, 2022-10-25:
Thanks for the update and explanation!