jedisct1/libhydrogen #141: Is there a way to do signcryption with this library?

creationix, 2023-03-14:
I really like the minimalism of this library for embedded use cases.  

I'm designing a protocol where I want authenticated messages stored in relays to be encrypted at rest and only readable if you know the public key.

The relay will index the messages by a hash of the public key (similar to how dat works) so that the relay never even knows the public keys and can never decrypt the messages.  But any client that knows the public key can ask for the messages from the relay and decrypt and authenticate them.

Is encryption alone enough to authenticate or is something else needed?

jedisct1, 2023-03-14:
If the public key of the recipient is not known, the shared secret cannot be recovered.

Signcryption doesn't seem necessary here, unless the relay needs to verify the sender's identity without being able to decrypt the content.

creationix, 2023-05-16:
The relay doesn't need to verify anything.  Maybe I misunderstood what signcryption does.

What I want is a system where:
- A message is encrypted end-to-end (relays can't read it)
- The sender doesn't know any keys of the receiver.
- The receiver only knows the sender's public key
- The relay doesn't know the sender's public key, only a hash of it (it's semi public)

My construction idea is:
- Sender encrypts a message using their private key (similar to signing, but the whole message is encrypted not just the content digest)
- how do I encrypt the bulk message using a private key?  I assume I want to embed some kind of symmetric key that is encrypted right?

jedisct1, 2023-05-16:
> The sender doesn't know any keys of the receiver.
> Sender encrypts a message using their private key

In that case only the sender can decrypt their own message. The recipient cannot do anything with it.
Is it what you want?

creationix, 2023-05-16:
The receiver knows the sender's public key.  Isn't that enough to decrypt the message?

creationix, 2023-05-16:
I think I'm misunderstanding the primitives.  [This clears it up for me a bit](https://crypto.stackexchange.com/questions/83266/why-are-ed25519-keys-not-recommended-for-encryption)

creationix, 2023-05-16:
I think ECIES might be what I'm looking for.  But I also saw that maybe I shouldn't use the same keypair for that as used for signatures.

creationix, 2023-05-17:
Or what I want without specifying implementation details:

- sender encrypts a message and publishes it publicly without disclosing the corresponding public key, only a hash of the public key
- authorized reads know from a side-channel how to get the public key from the hash of the public key and only they can decrypt it.

Is ECIES the correct construction for this using ECC?  Is this possible with this library?

closed 2025-02-11 by github-actions[bot]