jedisct1/libhydrogen #152: Is the text in the footer of all the Wiki pages intentional or vandalism?

CharlesScoville, 2024-05-05:
As stated in the title, most (all?) of the pages of the wiki have at the bottom the text "drii10291." I did some internet sleuthing and I found a few random accounts, a "Blocked" Git**lab** user, and ultimately linked to an email that is associated to the Github entity "Purity-Corner" and the Gitlab user "uzkjzjjz" ... which is where I stopped.

Can anyone confirm that this is intended/expected/legitimate ... as it looks more than a little sus to me. :/

CharlesScoville, 2024-05-05:
... no reply... but it looks to be gone now?

Not sure what the deal was, but I'm going to close this as solved/nonissue.

closed 2024-05-05 by CharlesScoville

jedisct1, 2024-05-06:
Hi!

It was indeed vandalism.

libhydrogen's wiki was open to contributions, and until now, these contributions were always good.

But it was recently vandalized by a shady account. Not sure what the intent was. Looks like some political messages in arabic.

Anyway, all their changes have been reverted, and I will unfortunately have to restrict access to the wiki.