jedisct1/libhydrogen #154: Cannot use MingW with the standard Makefile.

TWLemontree, 2024-08-05:
I download "winlibs-x86_64-posix-seh-gcc-14.1.0-llvm-18.1.8-mingw-w64ucrt-12.0.0-r3" from "https://winlibs.com/" and use cmd "mingw32-make.exe" to build libhydrogen, something wrong happened.

> mingw32-make.exe

cc -Os -mtune=native -fno-exceptions -Wall -Wextra -Wmissing-prototypes -Wdiv-by-zero -Wbad-function-cast -Wcast-align -Wcast-qual -Wfloat-equal -Wmissing-declarations -Wnested-externs -Wno-unknown-pragmas -Wpointer-arith -Wredundant-decls -Wstrict-prototypes -Wswitch-enum -Wno-type-limits -I.   -c -o hydrogen.o hydrogen.c
process_begin: CreateProcess(NULL, cc -Os -mtune=native -fno-exceptions -Wall -Wextra -Wmissing-prototypes -Wdiv-by-zero -Wbad-function-cast -Wcast-align -Wcast-qual -Wfloat-equal -Wmissing-declarations -Wnested-externs -Wno-unknown-pragmas -Wpointer-arith -Wredundant-decls -Wstrict-prototypes -Wswitch-enum -Wno-type-limits -I. -c -o hydrogen.o hydrogen.c, ...) failed.
make (e=2): 系统找不到指定的文件。
mingw32-make: *** [<builtin>: hydrogen.o] Error 2

jedisct1, 2024-08-05:
Maybe some files required by the compiler are missing on your system.

Try [installing zig](https://ziglang.org/download/) instead and type

`zig build -Drelease`

in the root folder of libhydrogen. You should then get the library in the `zig-out/lib/` folder.

jedisct1, 2024-08-05:
My guess is that you don't have commands such as `rm` and `install`, that are part of MSYS2.

The toolchain you installed is not going to be sufficient for most makefiles.

So yes, if you don't want to install MSYS2, the best way is to use zig instead.

closed 2024-08-05 by jedisct1